jeffbass/imagezmq #3: Show the live image from Raspberry Pi to Flask server

dakata12345, 2019-04-20:
Hello , I have this code for showing frame by frame videofeed from Raspberry Pi  but it doesn't work,and I don't understand why ,  can you help me please?
`@app.route('/video_feed')
def video_feed():
    """Video streaming route. Put this in the src attribute of an img tag."""
    return Response(gen_frames(),
                    mimetype='multipart/x-mixed-replace; boundary=frame')

def gen_frames():  # generate frame by frame from camera
    imageHub = imagezmq.ImageHub(open_port='tcp://*:3001')
    time.sleep(2)
    while True:
        # Capture frame-by-frame
        (rpiName, frame) = imageHub.recv_image()
        imageHub.send_reply(b'OK')
        yield (b'--frame\r\n'
           b'Content-Type: image/jpeg\r\n\r\n' + frame + b'\r\n')
`

jeffbass, 2019-04-20:
I have not tried using the imagezmq ImageHub class in a Flask application, so I cannot provide Flask specific help. But here are the things I suggest you check:
1. Make sure your Raspberry Pi sender is specifying the correct IP address AND port (3001 in your case) to point to your Flask server.
2. Make sure you start your Flask server and that it is running BEFORE you start the image sending client on the Raspberry Pi. The ImageHub server must always be running before a client is started.
3. Try changing the yield statement to a return statement. The fact that you are using a generator rather than a simple function may be having an impact on the ZMQ timing or control. ZMQ is written in C and may not be expecting the behavior of a generator rather than a function.
Let me know if any of these things helps!

dakata12345, 2019-04-20:
Thank you for your fast response !
I checked number 1 and 2 , and I've been doing it like so. I've tried using number 3 and this is what I'm getting 
`line 234, in gen_frames
    b'Content-Type: image/jpeg\r\n\r\n' + frame + b'\r\n')
TypeError: ufunc 'add' did not contain a loop with signature matching types dtype('S37') dtype('S37') dtype('S37')`
I really can't seem to understand why is this happening because it works very good with the example you provided..

jeffbass, 2019-04-20:
I took a few minutes to read the Flask documents. Flask is a web server, controlling a port to receive and respond to http style requests. imagezmq's ImageHub is also a server, but for receiving and responding to ZMQ style bytestring messages. I don't believe that the two can be used together. Either use Flask as the server or use imagezmq's ImageHub as the server.  I cannot see a way to use both of them together, since each of them works by waiting for, and responding to, incoming requests. Only one of them can control the incoming port. If you want to use Flask as your server for receiving images, it would probably be more suitable to use a package such as Requests (https://github.com/kennethreitz/requests) as your image sender. I don't believe imagezmq classes can be made compatible with Flask. Sorry.

closed 2019-07-16 by jeffbass