jeffbass/imagezmq #4: (python: num_process): GStreamer-CRITICAL **: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed

projectdrone2, 2019-04-30:
Jeff, all, since it's my 1st issue post on Github I hope I post it correctly without bothering you all.

When I run the sending part with a Logitech USB Cam, I get at "vs = VideoStream(src=0).start()" the error "(python: num_process): GStreamer-CRITICAL **: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed" and the process hang.

Then when I unplug the USB Cam, the process get out of hang state and return the following message:

VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
VIDEOIO ERROR: V4L: can't open camera by index 0
VIDIOC_REQBUFS: No such device
Traceback (most recent call last):
  File "test_2_rpi_send_images.py", line 30, in <module>
    sender.send_image(rpi_name, image)
  File "/home/pi/imagezmq.py", line 52, in send_image
    if image.flags['C_CONTIGUOUS']:
AttributeError: 'NoneType' object has no attribute 'flags'

Any clue where the issue is coming from?

jeffbass, 2019-05-02:
Congratulations on your 1st issue post on GitHub.  Well done. One suggestion: when posting code or tracebacks in GitHub issues, it is helpful to highlight your "code" portions of your issues post using the "insert code" editing tool at the top of the issue editing area. It looks like <>. You can select the portion of your issue post that is code or traceback messages, then click the <> symbol, and it will be shown in "code" format like this: 

```
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
VIDEOIO ERROR: V4L: can't open camera by index 0
VIDIOC_REQBUFS: No such device
Traceback (most recent call last):
File "test_2_rpi_send_images.py", line 30, in 
sender.send_image(rpi_name, image)
File "/home/pi/imagezmq.py", line 52, in send_image
if image.flags['C_CONTIGUOUS']:
AttributeError: 'NoneType' object has no attribute 'flags'
```

From your error, it looks to me like OpenCV is not successfully reading your Logitech USB cam. Before trying to get the USB cam working with imagezmq, I suggest you test the USB cam with a very simple Python program that just reads the camera frame and shows it with a cv2.imshow() window in a simple "while True" loop. Several things may be causing the issue. The messages:

```
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
VIDEOIO ERROR: V4L: can't open camera by index 0
```

make me suspect that your USB camera is not compatible with the OpenCV video camera function. I have had that problem with OpenCV myself. I was able to get a Logitech C920 USB cam to work with OpenCV just fine, but when I tried a Logitech C270 webcam, I got many different kinds of error messages. I was able to debug the problem by writing a simple program that just read the camera and displayed the images on the same computer. I hope that helps you solve the issue. Let me know if I can help further.

projectdrone2, 2019-05-02:
Thanks for taking time to give a feedback!
I carried on debugging and indeed it is an issue with Opencv on rpi since
1. the same USB cam works well with Opencv on my PC and
2. when I read pictures on the rpi instead of trying to capture picture from usb cam connected to the rpi then send from rpi and receive on PC with imagezmq just works fine

To make a long story short it's not an issue with imagezmq.
As a consequence, unless you object, I will close the issue.

jeffbass, 2019-05-02:
So glad you were able to isolate the problem!
Closing the issue.

closed 2019-05-02 by jeffbass