jeffbass/imagezmq #5: Question about MQTT

robmarkcole, 2019-05-24:
HI Jeff
this isn't an issue, but I am interested to know why you chose ZMQ when MQTT is very common for IOT? There are a few comments in [this article](https://www.pyimagesearch.com/2019/04/15/live-video-streaming-over-network-with-opencv-and-imagezmq/) but I am interested to hear your views.
Many thanks
Robin

jeffbass, 2019-05-28:
Hi Robin,
Question and answer discussions work fine as issues in GitHub. Easier than using the Wiki. 😎
MQTT was definitely one of the messaging protocols I looked at. There are 3 reasons I settled on ZMQ.
1. MQTT requires a broker server; ZMQ does not.
2. A broker server means transporting an image twice. Once from image source to message broker; a second time from the message broker to destination. ZMQ is peer to peer; images are transported one time only from my imagenodes to my imagehub. In most cases that means the image will pass over the network twice rather than just once. More on "broker vs brokerless" is here: http://zeromq.org/whitepapers:brokerless. 
3. The request-reply pattern in ZMQ allows me to have greater control over handling of non-replies and delays in sending images. The publish-subscribe pattern is available in both ZMQ and MQTT; I prefer the simple request-reply pattern of ZMQ. It's "faults" are actually "benefits" in my imagenode to imagehub system. There is a lot of discussion of the request reply pattern in the first couple of pages of: http://zguide.zeromq.org/page:all.
I did not do any careful timing tests to see whether ZMQ would be faster or slower than MQTT from end to end. But ZMQ is very, very fast in my imagehub to imagehub system.
I have been using ZMQ to run 21 imagenodes (raspberry pi's) with 2 imagehubs for over 2 years. It is fast, reliable and very simple. But MQTT would be a great choice also; it is a great library.

robmarkcole, 2019-05-29:
HI Jeff
many thanks for your detailed reply, you put a lot of thought and expertise into this project and it shows.
All the best
Robin

closed 2019-05-29 by robmarkcole