jeffbass/imagezmq #7: Fail to receive image on server, Works well on local machine

alter-sachin, 2019-06-03:
Hi ! This is not an issue related to your implementation.
I am looking to connect frames from my Picam to a google cloud based compute engine GPU

client running on pi : 
`import socket
import time
from imutils.video import VideoStream
import imagezmq
 
sender = imagezmq.ImageSender(connect_to='tcp://192.168.0.254:5555')  #for local machine, this works for me.
sender = imagezmq.ImageSender(connect_to='tcp://GPU_IP:9090')  #for the GPU , this does not receive frames..
rpi_name = socket.gethostname() # send RPi hostname with each image
picam = VideoStream(usePiCamera=True).start()
time.sleep(2.0)  # allow camera sensor to warm up
while True:  # send images as stream until Ctrl-C
	image = picam.read()
	sender.send_image(rpi_name, image)`

Server receiving frames on local machine & GPU have the same snippet running. 

`import cv2
import imagezmq
image_hub = imagezmq.ImageHub()
i = 0
while True:  # show streamed images until Ctrl-C
	i = i +1
	rpi_name, image = image_hub.recv_image()
	#cv2.imshow(rpi_name, image) # 1 window for each RPi
	#cv2.waitKey(1)
	#img = cv2.imread(image,0)
	cv2.imwrite("hello.png",image)
	image_hub.send_reply(b'OK')`

The server does not seem to receive the frames even though the local machine does receive the same frames.

I have checked that the ports are opened on the server side. Is there someway I could debug this issue ? I am using an office internet.
Would like to know your thoughts on how this could be debugged....
Thanks for your hard work with this repo :+1:

alter-sachin, 2019-06-03:
Server by default listens only at 5555. opened 5555 at the gpu end and works well. thanks

closed 2019-06-03 by alter-sachin

jeffbass, 2019-06-03:
@alter-sachin Glad you resolved the issue! I will use your issue to help me improve the documentation of imagzmq. Thanks for using it and helping me improve it.