jeffbass/imagezmq #8: Getting FPS of around 3

alter-sachin, 2019-06-06:
Hi, I have seen such behaviour where we get only 3frames per second over the network.
Is this normal behaviour ?
Any way I could pump this up ? Please let me know a direction for the same.

jeffbass, 2019-06-06:
Yes, low FPS can be normal depending on your application. A lot of things affect FPS when transmitting images over the network:
1. Image size
2. Image compression
3. Network loading; especially the number of RPi's transmitting images simultaneously
4. WiFi versus ethernet
The image size / compression factors have been the rate limiting items for me, but I always check all 4 of the above. I use relatively small images sizes (320 x 240 is my most common size when transmitting from Raspberry Pi). I also use jpeg compression (which cuts image size by 60-90%). You can see the jpeg compression example in the test program in this repository. I do not find the 320 x 240 image size limiting because I have use smaller sizes with most deep machine learning techniques downstream anyway. But if you need larger images sizes, network bandwidth and network load will be the dominant issues.

alter-sachin, 2019-06-07:
@jeffbass You are a very kind man indeed, will use these heuristics and run some tests. Thanks!

jeffbass, 2020-05-30:
You're welcome! I've added my comments above to a question in the FAQ, so I'm closing this issue.

closed 2020-05-30 by jeffbass