jeffbass/imagezmq #12: Cloud Google

ffabinhoo, 2019-07-19:
Hi Jeff,
I'm loving this libray and experiment something using opencv.
I'm able to run it locally, however, when I tried to connect an server on Google cloud receiving images from my local camera, the server is not able to receive those stream. I believe that I set up everything properly (firewall rules, etc) but the communication between the peers doesn't work.

Any ideas?
Regards,
Fabio

jeffbass, 2019-07-19:
Hi Fabio,
First, let me say I have not actually tried to use imagezmq on Google cloud, so I have no direct experience to offer you.
Some things you might want to check:
1. Are you trying to run one of the imagezmq test programs first? If so, which program pair are you running? Is it the same test program pair that you ran successfully locally?
2. I presume you running the hub / receive program on the Google Cloud. Is it giving any error messages? Or just silently doing nothing? 
3. Did you try running the first test program (the one that just sends and displays 2 different colored squares) on the Google Cloud instance? Do you have a way to run cv2.imshow() on your cloud instance?
4. The imagezmq port is 5555 by default. Of course, it needs to be the same on both sender and receiver. You can specify a different port if needed. The way you specify TCP address and port is shown in imagezmq [API docs.](https://github.com/jeffbass/imagezmq/blob/master/docs/api-examples.rst).
I would suggest trying Google searches for resources on StackExchange and other places for information on using ZMQ on Google Cloud. Someone has probably used ZMQ on Google Cloud before...or asked questions if they have not been successful. imagezmq is just a set of simple classes wrapping the ZMQ REQest/REPly protocol. How you specify ZMQ TCP address:port is not changed by imagezmq, so if anyone has gotten ZMQ to communicate with the Google cloud, that should be helpful info.

ffabinhoo, 2019-07-19:
Hi Jeff!
Thank for your quick replying!
I will run the test program first (As you mentioned - 3). If everything works fine, I will check the other tips.

Thanks a lot!
Regards,
Fabio

closed 2019-07-19 by ffabinhoo

ashutoshIITK, 2019-08-05:
Hi @ffabinhoo could you solve the problem? I am also in a similar situation. I want to send images to my AWS EC2 server. What things do I need to modify? Any leads would be appreciated.

ffabinhoo, 2019-08-07:
Yes, I could. However, I ended up deleting the code. The only think that I
did was to open the opencv output using <video>.

ashutoshIITK, 2019-08-07:
@ffabinhoo 
Thank you for your response. So you only deleted the client-side code and instead streamed the frames from RPi to Google Cloud? 
Did you still use the TCP server for sending streams?