jeffbass/imagezmq #13: Suggestion: Add PUB/SUB mode for sender/receiver

bigdaddymax, 2019-07-23:
Imagezmq is a pretty slick and robust library that helped me a lot with running live video streams from different webcams and other sources (not only Raspberry) to my home server.

In my project, I also needed to implement a monitoring functionality - I needed to create a socket I could occasionally connect to and receive a copy of a video stream that is transmitted across my application (for example, between preprocessor and object detecting module or between object detector and file writer).

With the current implementation of Imagezmq this is impossible - sending an image over REQ/REP socket is a blocking operation, therefore publishing side would freeze until receiving part is up and connected.

The solution was adding a non-blocking PUB/SUB mode - in this case, publishing side will not block and will continue to send images across application regardless of the state of the receiver. 

The cool thing about the client is that it can connect to multiple publishers so I could create a single client that monitors multiple cameras.

If this is something you'd like to add to this project I could make a respectful pull request.

jeffbass, 2019-07-23:
Hi! This sounds like a great idea and I think adding a PUB/SUB option would significantly enhance the imagezmq library.  I have a couple of requests that will make the merge easier for me: please include additions to the documentation and the API page. The existing readme.rst includes a section "Additional Documentation". Add a bullet there that adds a link "Using a PUB/SUB alternative to REQ/REP" and add a "pub-sub.rst" file in the docs folder. Also edit the api-examples.rst in the docs folder. 

Please feel free to submit the pull request in multiple stages to get feedback from me or feel free to use this thread for additional discussion.

Thanks for contributing,
Jeff

bigdaddymax, 2019-07-25:
Hi Jeff! I have created a pull request #14 

This is the first step, code only. 
I will try to add documentation, tests and examples pull requests as soon as possible.

This is my first experience in contributing to Python project so I would appreciate any comments and suggestions.

Thanks, 
Maksym

jeffbass, 2019-08-07:
Closing this issue; we are doing further discussion in your pull requests. 👍

closed 2019-08-07 by jeffbass