jeffbass/imagezmq #17: Sending images to a server not connected to same router

ashutoshIITK, 2019-08-07:
I am a bit new to networking. In the client-side, we specify the IP address and the port number. How can we send the images from client to a server or computer (Maybe EC2 or my personal laptop)?
What things are required to be taken care of? Any help will be greatly appreciated.

Thank you!

jeffbass, 2019-08-07:
I have not personally used imagezmq to send images to an IP address in the cloud, so I will let others speak to that. But here is some general guidance based on my own experience.

To have the client send to your personal laptop, you need to know your laptop's IP address -- as if you were using ssh to log into your laptop. That becomes the IP address you put into the client as it is done in these lines of [test_2_rpi_send_images.py:](https://github.com/jeffbass/imagezmq/blob/master/tests/test_2_rpi_send_images.py)

```
# use either of the formats below to specifiy address of display computer
# sender = imagezmq.ImageSender(connect_to='tcp://jeff-macbook:5555')
sender = imagezmq.ImageSender(connect_to='tcp://192.168.1.190:5555')

```
I have senders / clients and hubs / servers on 7 different routers around my farm and imagezmq works just fine whether they are on the same router or not. All the routers do have one "master DNS" router that makes sure all their TCP addresses are unique.

The trick to success is this: if you can ssh into a computer using its iP address, then you can use that same address as the IP address for imagezmq, as illustrated above.

The port number can be any port number that is free, as long it is specified as the same number for both sender and receiver.

For senders / clients and hubs / servers that are in different networks or in the cloud, it gets more complicated. You will have to consult with your internet service provider or you cloud service provider about obtaining an IP address and port that will work. If you can use their guidance to get ssh working to a cloud computer, then it is likely imagezmq will work the same way to the same computer. How various services handle opening ports is very specific to the service. I find that ssh is commonly on port 22, but sometimes on other ports (one of my servers uses port 2222 for ssh instead).

Use ssh as your test example to access your laptop or an EC2 instance. It will allow you to see if you have the correct IP address and if port 22 is open. Then find out the rules for getting some arbitrary port open (as in '5555', above). Then use imagezmq on that IP address and port.

I hope this helps. Every internet service provider, every external router and every cloud service has their own rules for determining a fixed IP address and for opening a port, so you will have to check with the service you will be using.

Good luck!

ashutoshIITK, 2019-08-07:
Dear Jeff,

Thank you so much for your quick response! Your comments are very helpful. I have understood the basic idea behind it. I will try it out.

Regards,
Ashutosh

closed 2019-08-09 by ashutoshIITK