jeffbass/imagezmq #18: Unable to establish the connection again after disconnection from server

ashutoshIITK, 2019-08-09:
Hi Jeff,

After establishing the connection between the server and client, when the client goes down (or I disconnect the client) while the server is running, I can re-establish the connection between the server and the client. When the same happens with the server (say I disconnect the server) and the client is running, I cannot establish the connection by turning the server up again. It does not receive any images.

What is the cause of this problem?

Thank you.

ashutoshIITK, 2019-08-09:
I solved the problem!

It was more related to how communication takes place between the client and the server in ZMQ framework. 
You can read more about it here: https://stackoverflow.com/questions/26915347/zeromq-reset-req-rep-socket-state

I reset the socket if there is a timeout (When no message is received from the server).

closed 2019-08-09 by ashutoshIITK

jcardenaslie, 2019-11-16:
Hi @ashutoshIITK is it possible that you could provide the piece of code that must be changed or a code snippet of how to do it. Thanks!

camilomarino, 2020-04-17:
I have the same problem, how do I solve it? What code snippet do you modify?