jeffbass/imagezmq #19: run test_2_mac_receive_images.py on Windows 7 , test_2_rpi_send_images.py on Raspberry Pi, server site no response

robertchou66, 2019-09-25:
Dear Jeff,

I encountered a problem.
I tried to run test_2_mac_receive_images.py on Windows 7 , test_2_rpi_send_images.py on Raspberry Pi, but server site no response.

on the Raspberry Pi site, ran "lsof |grep 5555"  will show a lots of these messages
python3   658 672 python3                 pi   14u     IPv4      16265      0t0        TCP 192.168.34.143:54214->192.168.34.93:5555 (SYN_SENT)
python3   658 673 python3                 pi   14u     IPv4      16265      0t0        TCP 192.168.34.143:54214->192.168.34.93:5555 (SYN_SENT)
python3   658 674 python3                 pi   14u     IPv4      16265      0t0        TCP 192.168.34.143:54214->192.168.34.93:5555 (SYN_SENT)
python3   658 675 python3                 pi   14u     IPv4      16265      0t0        TCP 192.168.34.143:54214->192.168.34.93:5555 (SYN_SENT)

on Windows 7 site, ran "netstat -an| grep 5555" , just showed only 
  TCP    0.0.0.0:5555           0.0.0.0:0              LISTENING

Can you tell what is these problem?

robertchou66, 2019-09-25:
I found out that Anaconda will add an InBound block rule for Python with connections from public networks.

That's the root cause.

jeffbass, 2019-09-26:
I don't use either Windows or Anaconda, so I'm afraid I can't be much help in resolving this. But please continue to add anything you learn if you come up with a solution that works for you. Thanks.

robertchou66, 2019-09-27:
Dear Jeff,
Yes, the solution is to change the inbound block rule to let any connections can go through the port 5555.
After that, everything will be fine.

closed 2019-09-27 by robertchou66