jeffbass/imagezmq #20: Does it work in Windows ?

siva2k16, 2019-10-24:
Please confirm if this works in Windows 10 too ?

jeffbass, 2019-10-24:
Another user asked about this in issue #19. I don't use Windows (or Anaconda, also mentioned in #19 ), so I'm afraid I don't know from my own direct experience. But please add anything to this issue if you come up with a test that gets any of the imagezmq tests working on Windows. Thanks.

jeffbass, 2019-10-24:
In issue #19, the coder mentioned that "the solution is to change the inbound block rule to let any connections can go through the port 5555". I don't know any more about imagezmq on windows than that.

siva2k16, 2019-10-25:
Okay Thanks jeffbass. Will try and update

hjinlee88, 2019-11-18:
I tested on python 3.6 in conda env on windows 10, both REQ/REP and PUB/SUB works.
http:// and localhost:port do not work but tcp:// and ip:port (such as 127.0.0.10:11000) works.

jeffbass, 2019-11-18:
Thanks, @hjinlee88 . I realy appreciate your testing.

AgardnerAU, 2020-03-09:
> I tested on python 3.6 in conda env on windows 10, both REQ/REP and PUB/SUB works.
> http:// and localhost:port do not work but tcp:// and ip:port (such as 127.0.0.10:11000) works.

Hi @hjinlee88, as a newbie, I've been able to get the test code on the pi send and receive, but am really struggling to receive any files on the windows machine. Would it be possible for you to post the code you used for both pi and windows 10 (redacted for IP if needed) so I can reverse engineer it?

Apologies if this is not the right form to ask for help.

@jeffbass Amazing work by the way.

jeffbass, 2020-03-09:
Hi @AgardnerAU , this is the best forum for your questions and requests for help. Since I don't use Windows or have any recent Windows experience, I can't help you directly. By posting your question here (on this issue / question), someone who is successfully using imagezmq with Windows might be able to help you. It was a good idea to direct your question to hjinlee88, who posted it works. I hope that you get a response from hijinlee88 in this thread. I will post a request in this thread specifically naming all the users who have asked questions about windows and hope to get a response.

jeffbass, 2020-03-09:
@AgardnerAU , here are the things I would try if I had a Windows computer. Please try them in order and report back by replying to this issue. Thanks!

1. Make sure you know your explicit localhost ip address on your Windows computer and that you can "ping" it OK:

```bash
# run the ping command on your Windows computer to check explicit localhost address
# use your Windows computer's localhost ip address
ping -c 5 127.0.0.1  
```

2. Install imagezmq on your Windows computer including all its dependencies needed for running the tests.

3. Run the pair of Test 1 programs on your Windows computer. Modify both test programs to explicitly specify the ip address and port to use by specifying the full localhost ip address:

```python
# in the test 1 receiving program running on your Windows computer:
image_hub  =  imagezmq.ImageHub(open_port='tcp://127.0.0.1:5555')  # use the same address

# in the test 1 sending program running on your Windows computer:
sender  =  imagezmq.ImageSender(connect_to='tcp://127.0.0.1:5555')  # use your localhost address
# The OpenCV imshow() window should open and you should see the frame count displayed.
```

4. If you get Step 3 working, it means imagezmq and all its dependencies are working OK. It also means that you can successfully specify your localhost address on your Windows computer.  It also means that you can use port 5555 locally on your Windows computer. Next, ping your Windows computer from your raspberry pi computer to make sure your raspberry pi can successfully reach your Windows computer's ip address. You will need to know your Windows computer's ip address:

```bash
# run the ping command on your raspberry pi to check network connection
# use your Windows computer's ip address
ping -c 5 192.168.86.39  # example ip address used with my Linux computers
```

5. Next, run the test 1 programs with your Windows computer receiving and your raspberry pi computer sending. You should specify both ip addresses explicitly:

```python
# in the test 1 receiving program running on your Windows computer:
image_hub  =  imagezmq.ImageHub(open_port='tcp://127.0.0.1:5555')  # keep same address as in Step 2

# in the test 1 sending program running on your raspberry pi computer:
sender  =  imagezmq.ImageSender(connect_to='tcp://192.168.1.39:5555')  # use your Windows computer ip address
# Use the same address that worked with  "ping" above
# The OpenCV imshow() window should open and you should see the frame count displayed.
```

6. If you try Step 5 and it hangs, you likely need to find a way to open the port on your Windows firewall.  Or, for Windows, you may need to use some port other than 5555. I cannot help you with those things on Windows. I am hoping someone else using Windows will offer a specific firewall or port suggestion in this issue. I have tested many combinations of raspberry pi sending and Mac or Linux computer receiving. In every combination, I have been able to get Steps 1 through 5 working OK. 

7. If Step 5 works, try the Test 2 programs using a raspberry pi sending images and your Windows computer receiving the images using the explicitly specified ip addresses as in Step 3. It should work if Step 5 did.  If not you probably need to troubleshoot your raspberry pi camera. I have tested many combinations of raspberry pi sending and Mac or Linux computer receiving. In every combination, when Steps 1 through 5 were working OK,  I have been able to get Step 6 working OK by debugging the raspberry pi camera.

Try the above steps and reply to this comment to report if it worked or did not work. If it doesn't work, at what step in the sequence did it fail?

If you don't get it to work, I will ask for help from specific users in an additional post on this issue. We may get a response. Good luck!

AgardnerAU, 2020-03-10:
Excellent help and thanks for the very speedy response!

I can confirm that it definitely does work in windows. It was a rookie mistake and I had the connect_to='tcp://192.168.1.39:5555' command in the sender python script, but had not opened the port in the server script.

jeffbass, 2020-03-10:
@AgardnerAU , thanks for letting me know it works for you in Windows!  I have a question: did you need to do anything (other than run your python program with imagezmq port parameters) to change the firewall or enable the port in Windows? That is where I don't know anything about Windows. In other words, once you changed your program as you mentioned above, did you have to do anything else to make it work in Windows? Thanks in Advance, Jeff.

AgardnerAU, 2020-03-11:
There may be  a more secure way, but for it to work for me, I had to add an exception to allow Python to communicate to public networks in the windows firewall settings. Even though both machines are on my private network, just permitting private access does not appear to allow it to function.

bdoligale, 2020-03-25:
Thank you, thank you, thank you AgardnerAU!  I messed with this for three hours tonight trying to get it to work, and your suggestion to open Python to public networks did the trick.  I don't really understand why it's needing that, but it's the only thing that worked among the numerous things I tried!  Thanks again!!  -Bob

bdoligale, 2020-03-25:
BTW, I had created a special inbound rule in Windows Firewall to allow port 5555 to receive.  I just disabled that rule and it still works fine.  I'd feel safer if the port rule had worked without opening up Python, but for now, I'll use it like it is.  -Bob
PS - Thanks, Jeff for your awesome contributions with ImageZMQ!  Great library!

jeffbass, 2020-03-25:
@bdoligale , I am glad you got it to work. I am putting together a FAQ page for  the documentation and will include what I have learned from you and @AgardnerAU . What I love about GitHub is that the community finds fixes for issues and shares them. Thanks to both of you for sharing and verifying what worked.

The1CABAL, 2020-05-14:
> Excellent help and thanks for the very speedy response!
> 
> I can confirm that it definitely does work in windows. It was a rookie mistake and I had the connect_to='tcp://192.168.1.39:5555' command in the sender python script, but had not opened the port in the server script.

Hey @AgardnerAU I am having the same issue you were and I do not seem to be able to get it to work. It works if my Windows machine is sending to a Linux server, and it works if my Windows machine sends to itself. But it is NOT working when sending from Linux to Windows...

I tried the trouble shooting tips posted by @jeffbass and it is still hanging...

on the server side I presently have `imageHub = imagezmq.ImageHub(open_port='tcp://127.0.0.1:5555')` and I tried actually assigning the IP address it has. I am wondering what specifically you did server side to get it to work. Thanks in advance!

Ilias9807, 2021-02-03:
@The1CABAL I am facing the exact same problem as you do. I am able to use imagezmq to send a stream of images from my Windows PC to the RPi, as well as from PC to it's self. But I am unable to send the stream from the RPi to my PC even after trying all the sugestions present on this thread. Have you resolved your problem, if so how? Has anyone else any other suggestions that worked for them?