jeffbass/imagezmq #21: AttributeError: module 'imagezmq' has no attribute 'ImageSender'

nandemoi, 2019-11-01:
Sorry this must be pretty basic.
When I tried to run client.py I got
AttributeError: module 'imagezmq' has no attribute 'ImageSender'

closed 2019-11-01 by nandemoi

jeffbass, 2019-11-01:
Glad you found a work-around. Let me take a look at the link you provide and try to understand what the issue is. I'll post another comment in this thread in a few days after I have looked at it.

nandemoi, 2019-11-01:
Thank you. Great work. Thanks for sharing.

I found it down below says

import imagezmq
by
from imagezmq import imagezmq

that resolved it.

ycdaskin, 2021-02-02:
Still getting this error. What @nandemoi suggested did not work for me. Any solution for this?

jeffbass, 2021-02-02:
Hi @ycdaskin, 
If you are getting this error, it is very likely that **imagezmq** has not been imported correctly. To use **imagezmq**, you need to create and use a Python virtual environment. OpenCV needs to be installed in the same virtual environment. In the rest of this post, I am assuming you have a virtual environment that has OpenCV already installed in it. Here is how to make sure that **imagezmq** has been installed correctly into that virtual environment:

```bash
# run these 2 commands at your system prompt
workon py3cv4  # my environment is named py3cv4; use your own virtual environment name
pip install imagezmq
```

Then, run Python interactively and import imagezmq and create an instance of the ImageSender class. Make sure all this is working without errors:

```bash
(py3cv4) pi@rpi24:~ $ python
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
>>> import imagezmq
>>> imagezmq.__version__
'1.1.1'
>>> sender = imagezmq.ImageSender()  # if you get an error here, imagezmq is not installed correctly
>>> type(sender)
<class 'imagezmq.imagezmq.ImageSender'>
>>> exit()
(py3cv4) pi@rpi24:~ $ 
```

Be sure that any Python programs that use **imagezmq** are run in this same virtual environment.
I think that running the above commands should enable you to find and fix your error. Let me know if I can help further.
Jeff