jeffbass/imagezmq #22: How to stream two videos on one device

JCP13, 2019-11-04:
Hi,

Thank you for the great project. I would like to stream two videos from one pi device but I am not sure how to do that, any thoughts?

Thank you in advance.

jeffbass, 2019-11-05:
I have sent 2 image streams from the same RPi by using a PiCamera and a USB Webcam attached to the same RPi. 

The existing imagezmq test programs show how to send a single stream of images from a single PiCamera. If you want to send a 2nd video stream of images from the same RPi, you would have to use a USB Webcam plugged into one of the USB ports on the RPi. I have one RPi set up with 1 PiCamera and 1 USB WebCam (a Logitech C920 Webcam). To test the 2 camera RPi setup I  modified test_2_rpi_send_images.py. Running 2 cameras on the same RPi requires 2 unique names for the 2 streams, and requires 2 camera instantiations from VideoStream and 2 image reads and 2 image sends. Here are the changes I made to  ```test_2_rpi_send.py```:

```python
rpi_name = socket.gethostname()  # send RPi hostname with each image
picam_name = rpi_name + " PiCamera"  # this name will show on PiCamera images
webcam_name = rpi_name + " WebCam"  # this name will show on PiCamera images
picam = VideoStream(usePiCamera=True).start()
webcam = VideoStream(usePiCamera=False).start()
time.sleep(2.0)  # allow camera sensor to warm up
while True:  # send images as stream until Ctrl-C
    image = picam.read()
    sender.send_image(picam_name, image)
    image = webcam.read()
    sender.send_image(webcam_name, image)
```

The test_2_mac_receive_images.py program did not need any changes.
Hope that helps. Let me know if this answers your question.

JCP13, 2019-11-05:
Thank you Jeff, that worked wonderfully 👍

closed 2019-11-05 by JCP13

jeffbass, 2019-11-05:
You're welcome! Glad it worked for you.