jeffbass/imagezmq #23: copy flag not honored

tatome, 2019-11-08:
Just noticed while browsing the code: the `receive_*()` methods of `ImageHub` don't use their `copy` argument.

eg:
https://github.com/jeffbass/imagezmq/blob/258453be9d86d213b31d83dcbcfcc68f26198328/imagezmq/imagezmq.py#L281

jeffbass, 2019-11-13:
Actually, the copy parameter **is** used. It is passed through a couple of layers of inheritance: `self.zmq_socket.recv_jpg()` is a method that inherits from class `SerializingSocket(zmq.Socket)`. The `copy` parameter is passed to the `zmq_socket` instance and then passed to the `zmq.Socket` method `recv()`. See lines 370 to 389, especially line 387:

https://github.com/jeffbass/imagezmq/blob/258453be9d86d213b31d83dcbcfcc68f26198328/imagezmq/imagezmq.py#L387

`self.recv()` is a `zmq.Socket` method. It gets its `copy` parameter value passed in to it from line 281 that you quoted above.  The class inheritance is a bit layered.  I have found that drawing a full instance / inheritance diagram helps to follow it.

tatome, 2019-11-13:
What I mean is, methods `ImageHub.recv_jpg()` and `ImageHub.recv_image()` both have  a parameter named `copy` with a default of `False`:
https://github.com/jeffbass/imagezmq/blob/258453be9d86d213b31d83dcbcfcc68f26198328/imagezmq/imagezmq.py#L271

but they don't use that parameter.  Instead, they always call methods on `this.zmq_socket` with a constant parameter `copy=False`:
https://github.com/jeffbass/imagezmq/blob/258453be9d86d213b31d83dcbcfcc68f26198328/imagezmq/imagezmq.py#L281

jeffbass, 2019-11-13:
I apologize for misunderstanding your question. I used the `copy` parameter during testing and debugging imagezmq.  I left in the `self.zmq_socket.rec*()` methods in case I need to use it again. I may someday add the changes to the code to make it available as a parameter of the ImageHub class.

jeffbass, 2020-03-10:
Closing #23

closed 2020-03-10 by jeffbass