jeffbass/imagezmq #24: Compression

petertorelli, 2019-11-10:
It looks like the image format is up to the VideoCamera codec, and that ImageZMQ is simply passing the raw data, which may be JPEG, or may not, depends on if PiCam is used. Is this true? Would it be more succinct to refer to the image as a binary data rather than jpeg in the code, since I see nothing specific to images... or am I missing something? Thanks! Great library, btw.

jeffbass, 2019-11-13:
Good catch. You are not missing anything. Both the image and jpeg functions are just passing binary blobs. In the current imagezmq version, the calling code does any jpeg transformation. You can see examples in the test programs with *jpg* in their names. In a future version, the jpg compression will be an option that does that transformation in the imagezmq classes. The current setup is a placeholder as I work on the next version.

petertorelli, 2019-11-13:
Whew, ok thanks for confirming!

closed 2019-11-13 by petertorelli