jeffbass/imagezmq #25: AttributeError: module 'imagezmq' has no attribute 'ImageHub'

jcardenaslie, 2019-11-11:
Working on windows.

I put the repo inside envname/Lib/site-packages

my script does import imgzmq. But no ImageHub attribute is found.

Any Ideas?

jeffbass, 2019-11-13:
ImageHub is a class, not an attribute. Can you perhaps provide a code example that shows what's going on?

jcardenaslie, 2019-11-14:
Hi! 

I follower this blog instructions: 
[](https://kezunlin.me/post/b8847d9f/)

And got this error:
```
(venv) PS C:\Users\jquin\Desktop\SOSAFE\imagezmq> python .\tests\test_1_receive_images.py
Traceback (most recent call last):
  File ".\tests\test_1_receive_images.py", line 34, in <module>
    image_hub = imagezmq.ImageHub()
AttributeError: module 'imagezmq' has no attribute 'ImageHub'
(venv) PS C:\Users\jquin\Desktop\SOSAFE\imagezmq>
```

jeffbass, 2019-11-14:
It looks like a problem with your Python path (but I admit I never use Windows, so I am not familiar with it). Please try this:
1. Create a temporary directory and cd into it (I called mine tempimagezmq)
2. Activate your virtual environment
3. Copy imagezmq.py into that directory
4. Copy the "test_1" files into that directory; your directory will have imagezmq.py and the test programs in the same temporary directory:
```
(py3cv3) jeffbass@jeff-macbook ~/tempimagezmq $ ls -lth
total 64
-rwxr-xr-x@ 1 jeffbass  staff   1.4K Nov 14 09:06 test_1_sub.py
-rwxr-xr-x@ 1 jeffbass  staff   1.2K Nov 14 09:06 test_1_send_images.py
-rwxr-xr-x@ 1 jeffbass  staff   1.4K Nov 14 09:06 test_1_receive_images.py
-rwxr-xr-x@ 1 jeffbass  staff   1.1K Nov 14 09:06 test_1_pub.py
-rwxr-xr-x@ 1 jeffbass  staff    13K Nov 14 09:05 imagezmq.py
```

5. Run the pair of test programs "test_1_*" in that directory. 
If that works, imagezmq is working on your Windows machine and you have a path problem.

The simplest solution is to simply copy "imagezmq.py" to your working Python directory and try it again. The file is only 14 K and you could simply copy it to any program directory that uses imagezmq. Not ideal, but you could do it that way until you solve the Windows Python path problem you are having. I cannot help you with that. I don't use Windows at all. 
Good luck!
Jeff

jcardenaslie, 2019-11-14:
Thanks! 

I made it work!

I copy the imagezmq folder into the tests scripts and changed the import to

```py
import imagezmq.imagezmq as imagezmq
```

jeffbass, 2019-11-15:
Glad you got it to work! 😎

jeffbass, 2020-03-10:
Closing #25

closed 2020-03-10 by jeffbass