jeffbass/imagezmq #26: zmq.error.ZMQError: Permission denied on ImageSender(REQ_REP = False)

jcardenaslie, 2019-11-16:
Running on Windows with python 3.7.0

```py
sender = imagezmq.ImageSender(connect_to="tcp://{}:5555".format(args["server_ip"]), REQ_REP = False)
```

```sh
Traceback (most recent call last):
  File ".\client.py", line 128, in <module>
    sender = imagezmq.ImageSender(connect_to="tcp://{}:5555".format(args["server_ip"]), REQ_REP = False)
  File "C:\Users\jquin\Desktop\SOSAFE\SoSaFe Logic\surveillance_client\imagezmq\imagezmq.py", line 55, in __init__
    self.init_pubsub(connect_to)
  File "C:\Users\jquin\Desktop\SOSAFE\SoSaFe Logic\surveillance_client\imagezmq\imagezmq.py", line 77, in init_pubsub
    self.zmq_socket.bind(address)
  File "zmq\backend\cython\socket.pyx", line 550, in zmq.backend.cython.socket.Socket.bind
  File "zmq\backend\cython\checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Permission denied
```

jeffbass, 2019-11-17:
This looks like a socket permission error. It may also be related to connect vs. bind in the PUB/SUB protocol. I will need a few days to check into it. If you set the option REQ/REP=True (and make appropriate tcp address changes), does the code work OK?

jcardenaslie, 2019-11-18:
> This looks like a socket permission error. It may also be related to connect vs. bind in the PUB/SUB protocol. I will need a few days to check into it. If you set the option REQ/REP=True (and make appropriate tcp address changes), does the code work OK?

The same line of code works fine whit REQ_REP=True

Thanks for your help :)

jeffbass, 2019-11-20:
Tested this and it works OK for me. I did have an issue on my SUB image receiver; I needed to put the full tcp address in the receiver program (which isn't necessary in the REQ/REP pattern). See this line in tests/test_1_sub.py and verify you are using the full tcp address in the ImageHub() instiation:

https://github.com/jeffbass/imagezmq/blob/258453be9d86d213b31d83dcbcfcc68f26198328/tests/test_1_sub.py#L35

Also, I have found that using numeric tcp addresses rather than text string tcp addresses helps while debugging a new connection. For example, `(tcp://127.0.0.1:5555 rather than tcp://jeff-macbook:5555)`.

Also, make sure that you have specified `REQ_REP=False` on both image receiver program AND in image sender program.

jcardenaslie, 2019-11-20:
Hi Jeff. Thanks! I'll try it out and get back to you. 
Thanks again for taking the time to help me.

jeffbass, 2020-05-30:
You're welcome. Let me know if you have further questions.

closed 2020-05-30 by jeffbass