jeffbass/imagezmq #29: Use ZMQ on android along with ImageZmq on RPI

jogiji, 2019-12-20:
I would like to use ImageZmq on RPI with some opencv processing and then stream those images onto an android phone/tablet for further action.
Can I use a ZMQ library for android along with ImageZmq ?

jeffbass, 2019-12-20:
I don't believe ZMQ will be helpful. The ZMQ library would need to be running as part of an image receiving program on the android phone/tablet in order to receive the image stream. The android phone/tablet would need a fixed IP address for the ZMQ sender to send to.  I don't believe that either of those is possible. There are probably other ways to stream images from a Raspberry Pi to a smartphone. When I Googled "stream video from raspberry pi to android app" several promising threads showed up. Perhaps take a look at some of those. Good luck.

jeffbass, 2020-03-10:
Closing #29

closed 2020-03-10 by jeffbass