jeffbass/imagezmq #30: Upload to pypi.org?

timsears, 2020-01-03:
@jeffbass thanks for this very nice library! 
Worth uploading to pypi. Willing to add the necessary files, test, and make a pull request if you want, but the upload should probably come from you.

timsears, 2020-01-03:
just noticed  pr #28. significant overlap with that.

jeffbass, 2020-01-04:
Thanks, @timsears ! Uploading to pypi.org is definitely on my TODO list for 2020. I am just getting around to doing a careful read of #28 and I will get back to you on this issue if more help is needed. 
Jeff

jeffbass, 2020-03-10:
Hi @timsears,
I've uploaded imagezmq to PyPI and have been testing it with the help of others for about a month. All seems to be working well. As you can see from the setup.py file, I used a template called "setup.py (for humans)" to flesh out the smaller setup.py you provided. Thanks for your help in getting the pip install process started! Closing this issue.
Jeff

closed 2020-03-10 by jeffbass