jeffbass/imagezmq #33: Unable to send Reply back to Rpi from Host

vasu15, 2020-02-25:
Hi, I was unable to send a response back to Rpi using Imagehub.send_reply.

jeffbass, 2020-02-25:
The Mac (image receiving computer) always sends a response. My test programs don't show how to capture the Mac's response in the RPi sending program. To capture the response, you add a variable to receive the return value of the "sender.send_image()" function:

```python
rpi_name = socket.gethostname()  # send RPi hostname with each image
picam = VideoStream(usePiCamera=True).start()
time.sleep(2.0)  # allow camera sensor to warm up
while True:  # send images as stream until Ctrl-C
    image = picam.read()
    reply_from_mac = sender.send_image(rpi_name, image)   # captures the reply
    do_something_with_reply(reply_from_mac)
```

Above code snippet is from "test2_rpi_send_images.py"  in the tests folder. Adding reply capture to the  other test programs would be similar.
I think your question is a good one. I will add something to the documentation to discuss this.
Please let me know if above answers your question so I can close the issue.
Thanks!
Jeff

jeffbass, 2020-05-30:
I have updated the 2 programs `timing_send_images.py` and `timing_send_jpg_buf.py` to provide another example of how this is done. Let me know if you have any other questions.

closed 2020-05-30 by jeffbass