jeffbass/imagezmq #35: Client reconnection is not possible when after server goes offline

camilomarino, 2020-04-18:
Hello, when the server is turned off, the client cannot reconnect. I wanted to know if there is any way to tell the client that in case of not receiving the reply message (for example in 5 seconds) and try to reconnect.

jeffbass, 2020-04-18:
Thanks for raising this issue. This is a known "feature" of ZMQ, @camilomarino. It was actually **imageZMQ** issue: #1! "Hangs at...". The most elegant fix is a great one from Pat Ryan @youngsoul https://github.com/jeffbass/imagezmq/issues/1#issuecomment-489476219. Take a look at that that. It may work well for you.

In my own projects, I use a "`with Patience()`" try / except block like this one in my **imagenode** main loop: [ see imagenode.py](https://github.com/jeffbass/imagenode/blob/master/imagenode/imagenode.py) (see lines 33-46). 

The "hanging" of the REQ client when the REP server restarts is a known ZMQ "feature" and is there by design (so that a dropped REQ won't go unnoticed by the sender). I use this ZMQ feature as a part of my own [yin-yang-ranch](https://github.com/jeffbass/yin-yang-ranch) project design -- I want the RPi's to know if they need to deal with a non-responsive **imagehub**.

jeffbass, 2020-05-30:
Please the FAQ page in the documentation. It has a more complete discussion of this. See FAQ Question "Why does image sender need to restart when the image hub program restarts?"

closed 2020-05-30 by jeffbass