jeffbass/imagezmq #36: zmq.error.ZMQError : Address already in use

omartin2010, 2020-04-27:
Great job Jeff. Love this.
As I'm using this in my scenario, I may need to sometimes deallocate the streamer and reallocate it depending upon the circumstances. I get this failure:
```
Traceback (most recent call last):
  File "src/detector.py", line 718, in async_stream_video
    REQ_REP=False)
  File "/usr/local/lib/python3.6/dist-packages/imagezmq/imagezmq.py", line 53, in __init__
    self.init_pubsub(connect_to)
  File "/usr/local/lib/python3.6/dist-packages/imagezmq/imagezmq.py", line 75, in init_pubsub
    self.zmq_socket.bind(address)
  File "zmq/backend/cython/socket.pyx", line 550, in zmq.backend.cython.socket.Socket.bind
  File "zmq/backend/cython/checkrc.pxd", line 26, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Address already in use
```
when I run this portion of my code more than once :
```
while True:
    video_streamer = imagezmq.ImageSender(connect_to=f'tcp://*:{target_port}', REQ_REP=False)
    # Do some work...
    video_streamer = None
```
Whenever it hits the 2nd line another time, I get the traceback above. Any idea how to specifically free this (I'm not myself familiar with `pyzmq`). But I would have imagined some kind of a context management : 
```
with imagezmq.ImageSender(...) as sender:
   # do stuff
```
But I'm not seeing this in the source code, but if there's missing cleanup code, we would need to implement that too.
Thoughts? Did I miss something? Is there an easy way to clean up sockets from this object?
thanks!

omartin2010, 2020-04-27:
Fixed my own issue:
```
video_streamer.zmq_socket.close()
```
Hope this can help someone. I *think* this is enough to make it work well, but I've not tested extensively. Enough for my project to work.

jeffbass, 2020-04-27:
Hi @omartin2010,
I'm glad you found the workaround. Closing the socket is exactly what is needed to allow a 2nd instantiation of `imagezmq.ImageSender()`.

You have found a definite bug in imageZMQ. My own use cases and testing never needed to re-instantiate the `ImageSender()`. There is, of course, a similar fault in ImageHub(). Thanks for submitting this issue. 

I will fix this bug by adding the appropriate `def __exit__(self, *other_stuff)` that includes `zmq_socket_close()`.  It will take me a few days, since I need to implement it and test it in a significant number of test programs.

In the meantime, your fix should work fine. I will leave this issue open in case someone else runs into this problem and needs your workaround.

Thanks again for finding this bug AND for providing the solution.
Jeff

jeffbass, 2020-05-22:
Hi @omartin2010 ,
I have fixed the use of imageZMQ classes in `with` statements. The newest version here on GitHub should fix your original problem. I also added a close method in version 1.1.0.

Could you test that this newest version on GitHub fixes your `with` statement issue? You would need to test from the GitHub repository. I have not update the version of imageZMQ on PyPI yet pending further testing.

Thanks,
Jeff

jeffbass, 2020-05-30:
Thanks again for finding this issue. It's fixed. Let me know if you have any further problems with it.

closed 2020-05-30 by jeffbass

omartin2010, 2020-05-30:
will do - for now I'm working on another section of my project, but I will get back to the one where I use the streaming soon, and I'll update the code to use the with statement versus deleting the object. Thanks for the fix!