jeffbass/imagezmq #37: Missing files in sdist

thatch, 2020-05-21:
It appears that the manifest is missing at least one file necessary to build
from the sdist for version 1.0.1. You're in good company, about 5% of other
projects updated in the last year are also missing files.

```
+ /tmp/venv/bin/pip3 wheel --no-binary imagezmq -w /tmp/ext imagezmq==1.0.1
Looking in indexes: http://10.10.0.139:9191/root/pypi/+simple/
Collecting imagezmq==1.0.1
  Downloading http://10.10.0.139:9191/root/pypi/%2Bf/97e/3368e445993b2/imagezmq-1.0.1.tar.gz (14 kB)
    ERROR: Command errored out with exit status 1:
     command: /tmp/venv/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-f0tgka7t/imagezmq/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-f0tgka7t/imagezmq/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-wheel-f0tgka7t/imagezmq/pip-egg-info
         cwd: /tmp/pip-wheel-f0tgka7t/imagezmq/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-wheel-f0tgka7t/imagezmq/setup.py", line 21, in <module>
        with open('PyPI_README.rst', 'r') as f:
    FileNotFoundError: [Errno 2] No such file or directory: 'PyPI_README.rst'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

jeffbass, 2020-05-21:
Thanks for the catch. I'll update the manifest and resend to PyPI in the next day or so.

jeffbass, 2020-05-21:
@thatch, I updated MANIFEST.in so your error should not occur now. I tested it with:
```
git clone https://github.com/jeffbass/imagezmq
cd imagezmq/
python3 setup.py sdist
cd dist
pip3 install *tar*
```
The `pip3 install` ran without errors for me. Could you retry for yourself when you have time, please?
Thanks for finding this bug.
Jeff

thatch, 2020-05-22:
Yes, works now.

closed 2020-05-22 by thatch