jeffbass/imagezmq #38: Possible bug setting "connect_to='localhost':5555" on ImageSender

sfrmattos, 2020-06-18:
Python 3.8.2 (Win10 x64)
Imagezmq V 1.1.1
--
I noticed that when using:

`sender = imagezmq.ImageSender(connect_to='tcp://localhost:5555', REQ_REP=False)`

it fails with:

`zmq.error.ZMQError: No such device`.

With REQ_REP enabled, it works like a charm using "localhost":

`sender = imagezmq.ImageSender(connect_to='tcp://localhost:5555')`

Using `127.0.0.1:5555` or `*:5555` with PUB/SUB also runs with no problems.

** I'm not a developer, so I'm sorry if this kind of reporting is not appropriate.

jeffbass, 2020-06-19:
@sfrmattos,
This is exactly the place to report this. Thanks!
I don't use Windows, but I have had 2 other imageZMQ users report issues with Windows related to specifying the TCP address. I'm glad you found a workaround; I'd suggest just avoiding the "`localhost`" wording and just stick with `127.0.0.1:5555.` I think this might be a problem with the interface between ZMQ and Windows. imageZMQ just passes the tcp address to ZMQ, and doesn't do any modification of it. So I suspect it may be related to how ZMQ sets up a PUB socket on Windows. ZMQ sets up PUB/SUB socket pairs differently than REQ/REP pairs. That may account for why using "localhost" works for one and not the other. Thanks for reporting this. I will leave this issue open for a while to see if it sparks any comment from others. You may also want to read some of the comments on issue #20.

sfrmattos, 2020-06-20:
great! Thank you @jeffbass.
Next week I'll try to debug and figure it out where exactly this error occurs and report back,.