jeffbass/imagezmq #40: Converting to usb camera not pi camera

danammeansbear, 2020-07-22:
Hi everyone, im currently running into an issue where I cannot video more then just the first frame without the shown frame freezing and not responding. 
I get a still image and not a video. Is this because im using one machine? i currently am too broke to get a raspberry pi so should i use am emulator ? can raspberry pi emulators grab ahold of my pc camera as picamera or is it cap.cv2.VideoCapture(0)

"""
Created on Tue Jul 21 18:10:04 2020
computerreciever.py
@author: adam
"""
#this code is for a computer to host images sent from itself or another windows
#computer
import cv2
import imagezmq
image_hub = imagezmq.ImageHub(open_port='tcp://192.168.56.1:55555')
while True:
    rpi_name, frame = image_hub.recv_image()
    cv2.imshow(rpi_name frame)
    cv2.waitKey(1)
    image_hub.send_reply(b'OK')

"""
Created on Tue Jul 21 18:20:42 2020

@author: adam
"""
#this code lets you send video or images from the machines camera to host
import cv2
import time
import socket 
from imutils.video import VideoStream
import imagezmq

sender = imagezmq.ImageSender(connect_to='tcp://192.168.56.1:55555', REQ_REP=True)
# Open the device at the ID 0

cap = cv2.VideoCapture(0)
currentFrame = 0
#Check whether user selected camera is opened successfully.
if not (cap.isOpened()):
    print('Could not open video device')

rpi_name = socket.gethostname() # send RPi hostname with each image
#picam = VideoStream(usePiCamera=True).start()
#frame = VideoStream(usePiCamera=False).start()
time.sleep(2.0)  # allow camera sensor to warm up
while True:  # send images as stream until Ctrl-C
        ret, frame = cap.read()
        sender.send_image(rpi_name, frame)
        print(rpi_name)
        break

jeffbass, 2020-07-22:
Hi @danammeansbear,

I can't read your code because it isn't formatted and indented. I can't tell if your `break` statement is in your `while` loop.

Please edit your post above (the 3 dots in the upper right of your post above have an "edit" pulldown.
You can make your Python read correctly in your post by marking it as code. Please see the `syntax highlighting` this guide: [GitHub Markdown](https://guides.github.com/features/mastering-markdown/). While you are editing your post, you can click the "Preview" tab to see what it will look like. Then you can click the "Write" tab and edit as needed.

Once we can read your code we may be able to offer suggestions. 

This is an example of what your code should look like when formatted correctly (but I cannot tell where your indentions actually were; I made a guess):

```python
rpi_name = socket.gethostname() # send RPi hostname with each image
#picam = VideoStream(usePiCamera=True).start()
#frame = VideoStream(usePiCamera=False).start()
time.sleep(2.0) # allow camera sensor to warm up
while True: # send images as stream until Ctrl-C
    ret, frame = cap.read()
    sender.send_image(rpi_name, frame)
    print(rpi_name)
    break
```

danammeansbear, 2020-07-22:
oh god i didn't really notice how it posted, will edit!

danammeansbear, 2020-07-28:
here is my receiving 

```
import cv2
import imagezmq
image_hub = imagezmq.ImageHub(open_port='tcp://192.168.1.68:5555')
image_hub.connect('tcp://192.168.56.1:55555')
while True:
    rpi_name, frame = image_hub.recv_image()
    cv2.imshow(rpi_name, frame)
    cv2.waitKey(1)
    image_hub.send_reply(b'OK')
    

```

jeffbass, 2020-07-29:
Hi @danammeansbear, 
I see 3 issues with your receiver:
1. You did not put the `REQ_REP=False` parameter in your `image_hub = imagezmq...` line. If it is specified in your sender, it must also be specified in your receiver.
2. Your last statement `image_hub.send_reply(b'OK')` cannot be used with the PUB / SUB mode. It needs to be taken out.
3. When using the sender program and the receiver program on the same machine, I have to use` "*"` as the address on the sender and '`127.0.0.1`' as the address on the receiver. That is true on my Mac and may not be true on your computer.

I modified your programs per the 3 points above and got them to work on my Mac webcam displaying on the same Mac. They worked fine when I made the modifications. Here is my sender program that worked:

```python
# this is the PUB / broadcast / sender program
import cv2
import time
import socket
from imutils.video import VideoStream
import imagezmq

# sender = imagezmq.ImageSender(connect_to='tcp://192.168.56.1:55555', REQ_REP=False)
#  ^^^^ ---- I commented out your above statement and replaced it with below

# when using PUB/SUB, on sending computer, I specify *:5555 as address
sender = imagezmq.ImageSender('tcp://*:5555', REQ_REP=False)

# Open the device at the ID 0
cap = cv2.VideoCapture(0)
currentFrame = 0

#Check whether user selected camera is opened successfully.
if not (cap.isOpened()):
    print('Could not open video device')

# rpi_name = socket.gethostname() # send RPi hostname with each image
rpi_name = 'Test Stream'  # using this because I am NOT using RPi

#picam = VideoStream(usePiCamera=True).start()
#cap = VideoStream(usePiCamera=False).start()
#frame = VideoStream(usePiCamera=False).start()
time.sleep(2.0)  # allow camera sensor to warm up
while True:  # send images as stream until Ctrl-C
        return_code, frame = cap.read()  # cap.read returns *2* args!!
        sender.send_image(rpi_name, frame)
        # print(rpi_name)
        # break     <<----- I had to comment out this line
```

Here is the receiving program that worked for receiving and showing frames on the same Mac:

```python
import cv2
import imagezmq
# image_hub = imagezmq.ImageHub(open_port='tcp://192.168.1.68:5555')
# image_hub.connect('tcp://192.168.56.1:55555')
# ^^^^ I replaced your 2 lines above with the one below
#
# You did not specify REQ_REP=False...must be specified in BOTH sender & receiver
image_hub = imagezmq.ImageHub(open_port='tcp://127.0.0.1:5555', REQ_REP=False)
while True:
    rpi_name, frame = image_hub.recv_image()
    cv2.imshow(rpi_name, frame)
    cv2.waitKey(1)
    # image_hub.send_reply(b'OK')  <<-- do NOT use this with PUB / SUB mode
```

Give these things a try and let me know how it works out.
Jeff

danammeansbear, 2020-07-29:
works perfectly,but I still got alottt to learn

jeffbass, 2020-07-29:
Hi @danammeansbear, 
I'm glad it worked. Don't worry...we all have a lot to learn...especially me!
Jeff