jeffbass/imagezmq #42: Provision for authentication

assman, 2020-07-29:
Hi

Just wanted to check, is there any built in provision for authentication? Any advice on how to secure our setup in general? 

I would like to setup some form of basic authentication where the server only processes the frames received from a legitimate client that I have deployed.

jeffbass, 2020-07-29:
Hi @assman,
**imageZMQ** has no provision for authentication. It does not have any security provisions at all. I designed it and use it for sending images between Raspberry Pi computers and image hubs that are all behind a firewall. **ZMQ** does have a lot of security choices that will show up on a Google search. But I won't be adding any security to **imageZMQ**. You may want to fork this repository and add security features to it. Or it may be easier to start fresh using some of the existing **ZMQ** security libraries that are available.
If you do find a good solution for your use case, it would be great if you could share it back on this thread.
Thanks,
Jeff

assman, 2020-07-30:
Thanks Jeff! I'll be sure to post here if I find something useful regarding authentication.