jeffbass/imagezmq #44: Is PUB/SUB pattern compatible with send_jpg() ?

ZhengPan2014, 2020-08-12:
@jeffbass Need your help here.
I got errors while trying to transport pictures in the format of **jpg** with message pattern **PUB/SUB**.
```
Traceback (most recent call last):
  File "./send.py", line 9, in <module>
    sender = imagezmq.ImageSender(connect_to='tcp://58.87.115.128:5555',REQ_REP=False)
  File "/usr/local/lib/python2.7/dist-packages/imagezmq-1.1.1-py2.7.egg/imagezmq/imagezmq.py", line 53, in __init__
    self.init_pubsub(connect_to)
  File "/usr/local/lib/python2.7/dist-packages/imagezmq-1.1.1-py2.7.egg/imagezmq/imagezmq.py", line 75, in init_pubsub
    self.zmq_socket.bind(address)
  File "zmq/backend/cython/socket.pyx", line 550, in zmq.backend.cython.socket.Socket.bind
  File "zmq/backend/cython/checkrc.pxd", line 26, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Cannot assign requested address
```
**sender.py** is as follow(**X.X.X.X** is ip of server):
```
#!/usr/bin/python
import sys
import socket
import time
import cv2
import imagezmq

# use either of the formats below to specifiy address of display computer
sender = imagezmq.ImageSender(connect_to='tcp://**X.X.X.X**:5555',REQ_REP=False)
# sender = imagezmq.ImageSender(connect_to='tcp://192.168.1.190:5555')

rpi_name = socket.gethostname()  # send RPi hostname with each image
cap = cv2.VideoCapture(0)
time.sleep(2.0)  # allow camera sensor to warm up
jpeg_quality = 60  # 0 to 100, higher is better quality, 95 is cv2 default
while True:  # send images as stream until Ctrl-C
    ret,image = cap.read()
    ret_code, jpg_buffer = cv2.imencode(
        ".jpg", image, [int(cv2.IMWRITE_JPEG_QUALITY), jpeg_quality])
    sender.send_jpg(rpi_name, jpg_buffer)
    time.sleep(1)
```

jeffbass, 2020-08-12:
Hi @ZhengPan2014, 
Yes, PUB/SUB works well with `send_jpg()`. Your sender code is the PUB (publisher), so it needs a TCP address like this:
 ```python
sender = imagezmq.ImageSender(connect_to='tcp://*:5555',REQ_REP=False)
```
The point is that your sender must have `"*" `as its TCP address. (or, if `"*"` won't work, then try `"127.0.0.1"`). When using PUB/SUB, it is the **receiver** that must specify a **numeric** tcp address. Comparing receiver and sender addresses in PUB/SUB:
```python
# PUB/SUB example:
    # on image sending computer:
    image_sender = imagezmq.ImageSender(connect_to='tcp://*:5555', REQ_REP=False)  # or can be connect_to='tcp://127.0.0.1:5555’

    # on the image receiving computer:
    image_hub = imagezmq.ImageHub(open_port='tcp://172.217.14.78:5555', REQ_REP=False)  # must specify the address for first sending computer
```

The docs in this **imageZMQ** repository contain an example of PUB/SUB with send_jpg:

- Sender (PUB) code is [here](https://github.com/jeffbass/imagezmq/blob/master/examples/pub_sub_broadcast.py).
- Receiver (SUB) code is [here.](https://github.com/jeffbass/imagezmq/blob/master/examples/pub_sub_receive.py)
- Docs for the PUB/SUB example are [here](https://github.com/jeffbass/imagezmq/blob/master/docs/fast-pub-sub.rst).

For a more complete understanding of the TCP addressing of senders versus receivers in PUB/SUB versus REQ/REP, see my reply in issue #43. 
Let me know if this works for you,
Jeff

ZhengPan2014, 2020-08-18:
> Hi @ZhengPan2014,
> Yes, PUB/SUB works well with `send_jpg()`. Your sender code is the PUB (publisher), so it needs a TCP address like this:
>
> ```python
> sender = [...]
@jeffbass  Yes, it works as what you said.
But I got another problem. As you know, I try to transport image over the network with imagezmq(long distance in geography location ). time delay is too havily to accept. I need to find the problem causing time delay. Is there any tool to help me to analyze which parts cause it?

jeffbass, 2020-08-20:
Hi @ZhengPan2014,
Sorry, I don't have any expertise in network timing or optimization, so I can't be any help with that.
I'll leave this issue open in case someone has a suggestion.
Good luck,
Jeff