jeffbass/imagezmq #52: Problem running on FileVideoStream

ilu3000, 2020-12-01:
Hi, I'm having problem with displaying the video onto another machine. I tried to modify the code in test_2_rpi_send_images.py into something like this:

sender = imagezmq.ImageSender(connect_to='my ip')
rpi_name = socket.gethostname() 
vid = FileVideoStream(path ='D:/Hrnet/ufc.gif').start()
time.sleep(2.0)  # allow camera sensor to warm up
while True:  # send images as stream until Ctrl-C
    vid_read = vid.read()
    sender.send_image(rpi_name, vid_read)

After I run the code, it can only display like 2 second of the video before it got flagged.

Traceback (most recent call last):
  File "test_2_rpi_send_images.py", line 32, in <module>
    sender.send_image(rpi_name, vid_read)
  File "C:\Users\Bolt\Anaconda3\lib\site-packages\imagezmq\imagezmq.py", line 106, in send_image_reqrep
    if image.flags['C_CONTIGUOUS']:
AttributeError: 'NoneType' object has no attribute 'flags'

Can you help me on solving this issue?

jeffbass, 2020-12-01:
I suspect that your program read all the frames and that the error occurred when sending an empty frame. Are you sure you did not hit the end of the frames in the GIF? The `test_2_rpi_send_images.py` program does not have any "end of stream" test, because it was designed to send a never-ending stream of images from a camera. If you are using the `FileVideoStream()` method from `imutils`, you will need to add a test for `vid.stopped`. It might look like this:

```python
sender = imagezmq.ImageSender(connect_to='my ip')
rpi_name = socket.gethostname()
vid = FileVideoStream(path ='D:/Hrnet/ufc.gif').start()
time.sleep(2.0) # allow camera sensor to warm up
while True: # send images as stream until Ctrl-C
    vid_read = vid.read()
    if not vid.stopped:
        sender.send_image(rpi_name, vid_read)
    else:
        break  # end the while loop because there are no more images to read from the file

```

ilu3000, 2020-12-01:
Thank you, sir. You fixed my problem. Thank you so much😊 I'm truly grateful and appreciate for your response. Thank you.

closed 2020-12-01 by ilu3000

jeffbass, 2020-12-02:
You are very welcome. I'm glad it fixed your problem. I will add something about this to the docs to help others who may run into the same thing. Thanks for bringing it to my attention.