jeffbass/imagezmq #53: advanced_http.py not working

bharat8, 2020-12-05:
There is an issue of connection when running the above script. I was unable to receive the image stream even on localhost. Any help is much appreciated.

jeffbass, 2020-12-06:
Hi @bharat8,
There are 3 programs required to run the advanced_http.py program:
1. `advanced_send.py` -- creates images with a rectangle with numbers in it and sends these images to an ImageHub on port 5555. You must be sure the TCP address in line 6 points to the TCP address of the computer running program 2, below. During testing on a single machine, use either localhost or 127.0.0.1.
2. `advanced_pub.py` -- receives the images from above (via REQ/REP protocol) on port 5555 and then sends them via PUB/SUB via port 5566.  It sets up a default REQ/REP `ImageHub` to receive the images in line 12. The default is to receive the images on port 5555 from program above. It sets up the PUB/SUB `ImageSender` in line 15 to send images on port 5566. This line **must** specify the TCP address with a `"*"` as shown (because it is a PUB sender).
3. `advanced_http.py` -- receives images from `advanced_pub.py` on TCP address 127.0.0.1 port 5566 via PUB/SUB. You could also specify localhost and port 5566. The address and port are specified on line 8. This program sends images to a simple web page server running on port 4000, bia a yield statement on line 12.

It would be best to test all 3 programs running on the same machine to make sure all the TCP addresses and ports are working before testing on multiple machines. The programs should be started in the opposite order listed above. Start the web server program `advanced_http.py` first. Start the `advanced_pub.py` program second. Start the `advanced_send.py` third. You will also need to start a web browser pointing to 127.0.0.1:4000 so see the images.

Please look at the example programs and get them to run first on a single machine with no changes, if you can. Note that their are different port numbers for sending the image from program 1 to program 2 (5555) versus program 2 to program 3 (5566). Also note that the TCP address in line 15 of advanced_pub.py MUST be `"*"`.

The programs were tested first on a Linux system and then on a Mac. Are you using Linux? Or a Mac? I don't know if this set of example programs can run on Windows.
Jeff

bharat8, 2020-12-10:
Thanks a lot @jeffbass ! I have Raspberry pi set up as a web server but I am having in receiving of images. Can you suggest some kind of improvisations to receive the images in real time?

jeffbass, 2020-12-12:
Could you elaborate your question a bit? What are you experiencing? Is one of the programs slow or not working for you? Are you transferring the images in image form or converting them to jpg format? I have found jpg's to be transferred faster in some applications (but not all of them).