jeffbass/imagezmq #56: Improve jpg conversion speed a bit

fjolublar, 2021-02-08:
This is NOT an issue with the imagezmq library, but I wanted to express my opinion to improve some speed in the use cases where this library is used. 

Instead of using OpenCV to encode/decode video frames, it might be better to use another library:  `simplejpeg`. 
With some testing I did, the speed of jpg conversion was up 10 to 30% faster. In my case, I had to optimize every step of the transmission. 

I have no affiliation with the library, just want to spread some knowledge.

jeffbass, 2021-02-08:
Thanks @fjolublar! Any speed improvement suggestions are always very welcome.

Would you be interested in providing an example program pair that illustrates your use of `simplejpg`? You could start with 2 example programs that are already in the imageZMQ examples folder, duplicate them, and modify them to use `simplejpg`. A pair of examples programs that would be a good duplication starting point would be [with_imageHub.py](https://github.com/jeffbass/imagezmq/blob/master/examples/with_ImageHub.py) and [with_ImageSender.py](https://github.com/jeffbass/imagezmq/blob/master/examples/with_ImageSender.py). They are documented in [the imageZMQ examples docs.](https://github.com/jeffbass/imagezmq/blob/master/docs/examples.rst#example-of-using-imagezmq-in-a-context-manager-with-statement)  It would be helpful to name the example programs something like `faster_jpg_hub.py` and `faster_jpg_send.py`.  You could modify the example documentation page by duplicating and modifying the `with_ImageSender.py` section. If you don't have time to do this as a pull request, no worries. I'll put it on my "to do list". It's a really good idea.

fjolublar, 2021-02-09:
Of course. It`s a pleasure to contribute to this work.  I will work on it.

jeffbass, 2021-02-09:
Thanks again for your pull request. Merged it.

fjolublar, 2021-02-10:
Thank you for giving me the opportunity. These were my very first Pull Requests :)

shumwaymark, 2021-02-26:
I tried this in my project, but didn't go well, at least as a drop-in replacement. 

``` python
        if self.publish_cam:
            self._rate.update()
            #ret_code, jpg_buffer = cv2.imencode(".jpg", image, 
            #    [int(cv2.IMWRITE_JPEG_QUALITY), camera.jpeg_quality])
            jpg_buffer = simplejpeg.encode_jpeg(image, 
                quality=camera.jpeg_quality, 
                colorspace='BGR')
            Outpost.publisher.send_jpg(camera.text, jpg_buffer)
```
The call into `encode_jpeg` raised the following exception 

`ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject`

j3soon, 2025-02-27:
In case someone would like to refer to the PR, the link is here: https://github.com/jeffbass/imagezmq/pull/57