jeffbass/imagezmq #60: is it possible to directly send video clips?

MeixinZhu, 2021-06-03:
Thanks for this great repo! It works very nicely for sending images. 
I wonder whether it is possible to directly send video clips to remote servers? instead of sending frames to the server then combine them into a video.

jeffbass, 2021-06-04:
No, `imageZMQ` is not suitable for sending video clips that are NOT implemented by sending individual image frames. With `imageZMQ` each image frame is a distinct `ZMQ` message. True "video streaming" with multiframe compression (like that used in most video streaming codecs) won't work with `imageZMQ`.

MeixinZhu, 2021-06-07:
Hi Jeff, thanks for your reply. I got it. Do you know any other approaches for video streaming? Thanks!

timsu92, 2021-06-09:
You can use OpenCV to read a video into frames, and also use OpenCV to save them to a clip.

timsu92, 2021-06-10:
However, I don't think sending clips using this is ideal, because you need to decode and encode, which causes quality loss.
Instead, it would be better to transmit the original bytes using 0mq, the upstream of this good work.

jeffbass, 2021-06-10:
Hi @MeixinZhu,
As @timsu92 mentioned, you may want to build your own video streaming protocol using [ZeroMQ](https://zeromq.org/get-started/). If not, there are many video streaming protocols and applications to choose from. Which one to choose is really dependent on what you are trying to do. You might want to look at tutorials for [ffmpeg](https://opensource.com/article/17/6/ffmpeg-convert-media-file-formats) or [gstreamer](https://gstreamer.freedesktop.org/documentation/tutorials/index.html?gi-language=c). If you want to stream a webcam to a browser, you may want to start with one of these Webcam to Browser video streaming tutorials using Flask: [PyImageSearch](https://www.pyimagesearch.com/2019/09/02/opencv-stream-video-to-web-browser-html-page/) or [Miguel Grinberg](https://blog.miguelgrinberg.com/post/flask-video-streaming-revisited).

MeixinZhu, 2021-06-18:
Thank you!! This is really helpful.