jeffbass/imagezmq #62: Is PUB/SUB compatible with PiCam?

Kygru, 2021-06-29:
I am wondering if its possible to send PiCam with PUB/SUB rather than REQ/REP?

When attempting to use PUB/SUB with a PiCam I see the following error:
`Traceback (most recent call last):
  File "pub.py", line 18, in <module>
    sender.send_image(rpi_name, image)
  File "/home/pi/.virtualenvs/py3cv4/local/lib/python3.7/site-packages/imagezmq/imagezmq.py", line 129, in send_image_pubsub
    if image.flags['C_CONTIGUOUS']:
AttributeError: 'NoneType' object has no attribute 'flags'`

pub.py:
`import sys
import time
import numpy as np
import cv2
from imutils.video import VideoStream
import imagezmq
import socket

# Create an image sender in PUB/SUB (non-blocking) mode
sender = imagezmq.ImageSender(connect_to='tcp://*:5555', REQ_REP=False)

rpi_name = socket.gethostname()
picam = VideoStream(usePiCamera=True).start()

while True: 
    image = picam.read()
    sender.send_image(rpi_name, image)
    time.sleep(1)`

sub.py:
`import cv2
import imagezmq

image_hub = imagezmq.ImageHub(open_port='tcp://127.0.0.1:5555', REQ_REP=False)
while True:
    image_name, image = image_hub.recv_image()
    cv2.imshow(image_name, image)
    cv2.waitKey(1)`

Kygru, 2021-06-29:
Update - was possible after allowing sensor warmup with:
`time.sleep(2.0)`

closed 2021-06-29 by Kygru

jeffbass, 2021-06-29:
Glad you solved it. I have found that I even need `time.sleep(3.0)` on some Raspberry Pi computers. It seems to vary from computer to computer.