jeffbass/imagezmq #63: One source and multiple processors on same device ?

kpoman, 2021-07-13:
Hello,
I am curious on a simple setup where an RPi captures from a device and publishes to local processors (each would do a specific, independent task, say P1 detect face, P2 detect QRCode, etc.). As PUB/SUB uses a broadcast on a specified port, this wouldnt be the case as all processors are python processes running on the same host (each on their own cpu core for example). How can this be accomplished here ?
Thank you for any tip !

jeffbass, 2021-07-14:
Hi @kpoman,
You can run `imagezmq` senders and receivers on the same RPi computer. I do this a lot for testing. With PUB/SUB, you can have one Python program sending images and then run multiple Python programs receiving the images. You specify the same port (127.0.0.1) on every receiving (SUB) Python program. 

There is a pair of example PUB/SUB programs in the `imagezmq` **examples** folder:
Sender program (PUB): https://github.com/jeffbass/imagezmq/blob/master/examples/pub_sub_broadcast.py 
Receiver program (SUB): https://github.com/jeffbass/imagezmq/blob/master/examples/pub_sub_receive.py

You can have as many programs as you wish running as Receiver(SUB) on the same RPi. All of them must specify the same localhost (127.0.0.1) address and the same port number.

As a test, I started 3 terminal windows on a RPi that had a working camera module attached. I activated the same virtual environment in each terminal window. In the first terminal window, I ran `python pub_sub_broadcast.py`. In the other 2 terminal windows, I ran `python pub_sub_receive.py`. As expected, 2 windows opened and showed the images being sent from the camera. I did not need to modify the example programs in any way. They worked as is, with the addresses in all the example programs unchanged.

So, to start your experiments, try to repeat what I did and run the example programs in 3 terminal windows on the same RPi as I did. Let me know if that works for you. You could then create your own programs for receiving and processing images using the PUB/SUB example programs as a model.
Jeff

kpoman, 2021-07-17:
> Hi @kpoman,
> You can run `imagezmq` senders and receivers on the same RPi computer. I do this a lot for testing. With PUB/SUB, you can have one Python program sending images and then run multiple [...]

Awesome Jeff, exactly what I was imagining. I am trying to build a simple access control using license plate, face and fingerprint recognition with multiple cams on a single RPi3. The device is not very powerful but being able to do as much as possible on it (and with the help of a video stream broadcasted locally) + some computer vision processors on different cores, then, at the end, every detected event submitted via some messaging like RabbitMQ to ensure at least the delivery of the detected event (the crucial information).

kpoman, 2021-07-17:
> Awesome Jeff, exactly what I was imagining. I am trying to build a simple access control using license plate, face and fingerprint recognition with multiple cams on a single RPi3. The device is not [...]

Oh and btw, the cam able to do facial recognition should also be able to detect qrcodes and eventually (if the cam resolution is enough) read the fingerprint. I think I should do some tagging as soon as possible in order for the post-processors (finger, face landmarks, qrcode reader, etc.) to evaluate/trigger only interesting images. Maybe put some opencv on the pub side and add some tag as the message ? I mean, if I broadcast images + tags, any subscriber will have to retrieve the full stuff and use Pi ressources (even if I discard the image on processors not concerned by a specific tag) ?