jeffbass/imagezmq #69: Reading the last message in the queue with ImageHub

gabrielesartor, 2021-10-05:
Is there a way to read directly the last image in a queue with class ImageHub?
For example, I'm receiving continuously images on the TCP channel, but the connection goes down.
After 20 minutes the connection is restored and I want to receive the images from that moment and not the previous ones.

In other words, I'd like to have a behaviour similar to a receiver using the flag CONFLATE. An example below:

context = zmq.Context()
socket = context.socket(zmq.SUB)
socket.setsockopt(zmq.CONFLATE,1)
print("Collecting updates from weather server...")
socket.connect ("tcp://*:5555")

However, it seems that this flag cannot be used with images. Any suggestions?
Thank you in advance.

MichouTest, 2022-02-02:
Hello, I use this code to get last frame https://github.com/jeffbass/imagezmq/blob/master/examples/pub_sub_receive.py 
But if you restart the sub you may have some delay ...

jeffbass, 2022-02-02:
The above suggestion from @MichouTest is a good one and may solve your problem.

There are 2 other things that might be affecting this:

1. ZMQ options behave differently with multi-part messages. `imageZMQ` sends multi-part messages (the message part and the image part). There is a good explanation in this [Stackoverflow question](https://stackoverflow.com/questions/51380541/zmq-conflate-socket-option-cause-no-messages-to-be-received). If you read all the answers, there is also an explanation of ZMQ option setting. It may not be possible to use `imageZMQ` with the `zmq.CONFLATE` option because of this.
2. Most ZMQ options require that the context be closed, then the option set, then the context and socket re-opened in order for the option to be set correctly. That is not easy to do in `imageZMQ` and would probably require modification of the context  and socket setup in the imageZMQ classes. When I need to set ZMQ options, [this is how I do it](https://github.com/jeffbass/imagezmq/blob/master/examples/timeout_req_ImageSender.py). It works for me for some ZMQ options, including `zmq.RCVTIMEO` and `SNDTIMEO`. You may want to try this way of setting the `zmq.CONFLATE` option. I have never used the `zmq.CONFLATE` option, so I don't have suggestions of the best way to use it or if it would help your problem.