jeffbass/imagezmq #71: Converting minimal demo to Pub/Sub only transmits single frame.

mullenba, 2022-01-19:
When I convert the basic demo from the repo's main page to Pub/Sub, the receiving computer shows the first frame, then doesn't change.  The demo as written works fine.

I've tested that the loops are running, but the frame image doesn't change.

Sender:
```
import socket
import time
from imutils.video import VideoStream
from imagezmq import imagezmq

sender = imagezmq.ImageSender(connect_to='tcp://*:5555', REQ_REP=False)

rpi_name = socket.gethostname() # send RPi hostname with each image
picam = VideoStream(usePiCamera=True).start()
time.sleep(2.0)  # allow camera sensor to warm up

while True:  # send images as stream until Ctrl-C
   frame = picam.read()
   sender.send_image(rpi_name, frame)
```

Receiver:
```
import cv2
from imagezmq import imagezmq

image_hub = imagezmq.ImageHub(open_port='tcp://192.168.68.134:5555', REQ_REP=False)

while True:  # show streamed images until Ctrl-C
    rpi_name, image = image_hub.recv_image()
    cv2.imshow(rpi_name, image) # 1 window for each RPi
    cv2.waitKey(1)
```

jeffbass, 2022-01-19:
I don't see any issues with your code, but I would suggest you run the PUB / SUB example programs from the examples folder. Docs and links to code are [here](https://github.com/jeffbass/imagezmq/blob/master/docs/examples.rst#equivalent-pubsub-code-for-the-multiple-rpi-video-streaming-example).

Please try that and let me know what happens. I know that this pair of example programs works. Of course, you'll need to change your TCP address in the receiver appropriately.

mullenba, 2022-01-19:
The test_4 example works.

mullenba, 2022-01-19:
Doing more experimenting, it appears to be on the receive side.  If I start the sender, then start the receiver with nothing in front of the camera, I see the frame of the empty space.  If I then close the receiver and put my hand in front of the camera, restarting shows my hand.

I'm working with a Raspberry Pi and Windows desktop, and happens in both Windows and Ubuntu WSL.

mullenba, 2022-01-19:
Looks to be an issue with imutils VideoStream.  Replacing it with a direct call to picamera and sending that works fine.

closed 2022-01-19 by mullenba

jeffbass, 2022-01-20:
Thanks for the update. I have heard other reports of imageZMQ users resorting to direct calls to picamera after having issues with imutils VideoStream. For example see this issue an imagenode / imageZMQ user had with imutils VideoStream: [imagenode issue 15.](https://github.com/jeffbass/imagenode/issues/15)