jeffbass/imagezmq #72: UDP Support?

kevinlinxc, 2022-03-08:
Hi, I am using this library and have noticed that it is a bit slow for high frequency image sending, which is to be expected. I'm wondering if it's possible to use UDP instead of TCP? Doing a bit of research, it seems like ZMQ has some support for this with NORM and PGM:

http://wiki.zeromq.org/topics:norm-protocol-transport
https://stackoverflow.com/a/70623213/11741232

When I tried norm and pgm, I got zmq.error.ZMQError: Protocol not supported so it seems like it's not supported

jeffbass, 2022-03-10:
I have not used UDP, but `imageZMQ` uses ZMQ multi-part messages (using the ZMQ option `zmq.SNDMORE`). I believe this ZMQ option is incompatible with UDP. Also, with multi-part messages it is not likely using UPD would actually be faster; image messages are quite large, making the relative savings of UDP versus TCP relatively small. From my reading of the ZMQ docs, PGM may be compatible with ZMQ multi-part messages, but perhaps not with the pyZMQ bindings used by imageZMQ.

I saw that your post on stackoverflow got a number of responses. I would love to know what your learn from your experiments if you try some of the things mentioned there.

kevinlinxc, 2022-03-10:
Hi Jeff, that is good insight on UDP vs TCP being a small difference for large amounts of data, thank you. I'll keep you in the loop.