jeffbass/imagezmq #91: topics for broker usage

KobusG, 2024-07-12:
Does imagezmq support the use of topics? I see all the connections open with the empty topic string, and the data itself is in an array.
I would like to implement a broker strategy, where multiple producers can connect to a broker, and the broker then invokes a subscriber for each publisher, to listen to the given topic. 
 
I think the only way with imagezmq would be to have the broker physically receive all images, and open its own pipe to listeners. I'm concerned that would cause a lot more CPU usage than the typical zmq broker strategy. 

Otherwise I would have to have each producer use its own port, and send a sepperate signal via dbus or pipe or something to the coordinator to invoke the appropriate listener with that port...

jeffbass, 2024-07-12:
imageZMQ does not support topics or brokers. Each (text, image) tuple goes to a single hub, which uses the text portion of each tuple to categorize & store the image portion. The imageZMQ design is very simple and would not be easy to modify for your use case. 

ZMQ can be used to set up brokers and subscribers. There are descriptions of many suitable ZMQ messaging patterns in the ZMQ docs. You might want to look [here](https://zguide.zeromq.org/docs/chapter5/) to get started. If you design your own system using ZMQ (and PyZMQ), you may want to use the ``zmq.SNDMORE`` flag to combine a text message with an OpenCV image the same way imageZMQ does. Take a look at the imageZMQ code for REP/REQ sending to see one way to do that. Code is [here](https://github.com/jeffbass/imagezmq/blob/master/imagezmq/imagezmq.py), lines 345-366.

I think MQTT (docs [here](https://mqtt.org)) would be a good candidate for the system you describe. You may want to take a look at that. 

ROS has a messaging system that is very capable of complex patterns like the one you describe, but ROS is large and has a steep learning curve. Links to the ROS publish / subscribe broker system are [here.](http://wiki.ros.org/ROS/Concepts) 

If you find a good solution for your system, you may want to share your solution as a reply to this issue thread. There are probably other folks who are looking for something like you are. Your experience would be helpful.

KobusG, 2024-07-14:
Thanks for the reply. I'll come up with something else.

closed 2024-07-14 by KobusG