jeffbass/imagezmq #99: Non-blocking io

mamintatarin, 2025-07-21:
Hello! It would be nice to have something like .poll() to use it before .recv_image() in case I don't want to block my program forever (when I don't have the input stream). It's very useful with asyncio and the feature exists in PyZMQ, as far as I know. Thanks!

jeffbass, 2025-07-21:
Great suggestion. I will look into it and comment here with what I learn. I already use the ZMQ time out options (`zmq.RCVTIMEO` and `zmq.SNDTIMEO`)to raise an error when sending takes too long per this [example program](https://github.com/jeffbass/imagezmq/blob/master/examples/timeout_req_ImageSender.py). I don't know if it would help on the receiving side.

jeffbass, 2025-07-22:
Thanks, @shumwaymark! Very helpful.

jeffbass, 2025-07-22:
@mamintatarin, try Mark's code and see if it works for you. Looks good to me.